4IceG/luci-app-modemband #1: 大佬 我使用中兴 MF832s 4G卡托  貌似不兼容

firkerword, 2022-03-06:
插上之后识别是大容量存储，能识别出ttyUSB，但是过一会又消失了，用usb-modeswitch也不能切换模式，看了下脚本里面没找到这个型号，技术太差实在是搞不定，大佬可以指点一二吗

4IceG, 2022-03-06:
Hi,
First of all, please English (I am not fluent in Chinese).

Modem must have a port for communication, mostly it's any of ttyUSBX. If it doesn't have one, it won't be supported.

1) What is the modem, how does it appear in the system (Vendor, ProdID)?
2. Modem is in the supported list?

4IceG, 2022-03-06:
Look for a way to put the modem into the correct mode:

https://wiki.archlinux.org/title/ZTE_MF_823_(Megaphone_M100-3)_4G_Modem
https://gsmforum.ru/threads/zte-mf823-i-vsjo-chto-s-nim-svjazano.188925/#post-934477

I can't help you any more.

closed 2022-03-06 by 4IceG