4IceG/luci-app-modemband #2: Request add support modem

lrdrdn, 2022-03-08:
# Fibocom L850-GL
```
at+gmm
L850 LTE Module

OK
```

## Firmware version

 
```
AT+GTPKGVER?
+GTPKGVER: "18500.5001.00.05.27.16_5001.05.001.033"

OK

```

## vid & pid

8087:095a (NCM)
2cb7:0007 (MBIM)

### Serial port

/dev/ttyACM0 & /dev/ttyACM2

## All band supported

3G: 1,2,4,5,8
4G: 1,2,3,4,5,7,8,11,12,13,17,18,19,20,21,26,28,29,30,38,39,40,41,66

## Read active band

```
at+xact?
+XACT: 4,2,1,1,2,4,5,8,101,102,103,104,105,107,108,111,112,113,117,118,119,120,121,126,128,129,130,138,139,140,141,166

OK
```

## Lock 4g only B1

```
AT+XACT=2,,,101

OK
at+xact?
+XACT: 2,2,,1,2,4,5,8,101

OK
```

## Lock 4g only B1 & B3

```
AT+XACT=2,,,101,103

OK
at+xact?
+XACT: 2,2,,1,2,4,5,8,101,103

OK
```

## Lock All 4g Band

```
AT+XACT=2,,,0

OK
at+xact?
+XACT: 2,2,,1,2,4,5,8,101,102,103,104,105,107,108,112,113,114,117,118,119,120,125,126,128,129,130,132,138,139,140,141,142,146,166

OK
```

## Auto All band

```
AT+XACT=4,,,0

OK
at+xact?
+XACT: 4,2,1,1,2,4,5,8,101,102,103,104,105,107,108,112,113,114,117,118,119,120,125,126,128,129,130,132,138,139,140,141,142,146,166

OK
```

lrdrdn, 2022-03-08:
# Fibocom L860-GL
Basicaly same as fibocom L850-GL
```
at+gmm
L860 LTE Module

OK
```

# Firmware version

 
```
AT+GTPKGVER?
+GTPKGVER: "18600.5001.00.35.01.57_5026.01.001.005"

OK

```

## vid & pid

8087:095a

### Serial port

/dev/ttyACM0 & /dev/ttyACM2

## All band supported

3G: 1,2,4,5,8
4G: 1,2,3,4,5,7,8,12,13,14,17,18,19,20,25,26,28,29,30,32,38,39,40,41,42,46,66

```
at+xact?
+XACT: 4,2,1,1,2,4,5,8,101,102,103,104,105,107,108,112,113,114,117,118,119,120,125,126,128,129,130,132,138,139,140,141,142,146,166

OK
```

4IceG, 2022-03-08:
Hi,
Install the package / packages and copy the file from the attachment to "/
usr / share / modemband" and see if everything works.

If you are in mbim mode then of course rename the file.

lrdrdn, 2022-03-08:
> Hi, Install the package / packages and copy the file from the attachment to "/ usr / share / modemband" and see if everything works. If you are in mbim mode then of course rename the file. wt., 8 mar 2022, 08:44 użytkownik lrdrdn ***@***.***> napisał:
> […](#)

where I cand find the attachment?

4IceG, 2022-03-08:
Did you get a file called 8087095a?

lrdrdn, 2022-03-08:
no I can't find anywhere

4IceG, 2022-03-08:
Oh, I see that it cuts off everything that I added by mail, I can't see the
file, link to the file :|, I will send the file in the evening normally via
github, you will check it tomorrow.

4IceG, 2022-03-08:
OK, Install the package modemband / packages and copy the files from the attachment to "/usr/share/modemband" and see if everything works.
[8087095a.txt](https://github.com/4IceG/luci-app-modemband/files/8210062/8087095a.txt)
[2cb70007.txt](https://github.com/4IceG/luci-app-modemband/files/8210063/2cb70007.txt)

4IceG, 2022-03-09:
OK, copy the files from the attachment to "/usr/share/modemband" and see if everything work
[test2.zip](https://github.com/4IceG/luci-app-modemband/files/8213077/test2.zip)
s

4IceG, 2022-03-09:
OK, copy the files from the attachment to "/usr/share/modemband" and see if everything works
[test2.zip](https://github.com/4IceG/luci-app-modemband/files/8213215/test2.zip)
.

lrdrdn, 2022-03-09:
Foxconn T77W968  works fine. Another Foxconn modem

# Foxconn T77W767

vid & pid, It has 2 brand vendor HP & Telit
```
03f0:0a57 (HP lt4220 MBIM)
03f0:0857 (HP lt4220 QMI)
1bc7:1901 (Telit Ln940 MBIM)
1bc7:1900 (Telit Ln940 QMI) 
```
ATI
```
ati
Manufacturer: QUALCOMM INCORPORATED
Model: 4105
Revision: T77W676.F0.0.0.4.7.GC.017  1  [Jul 11 2018 00:00:00]
SVN: 03
IMEI: xxx
+GCAP: +CGSM,+DS,+ES
```

cat /sys/kernel/debug/usb/devices
```
	T: Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#= 7 Spd=480 MxCh= 0 	
	D: Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs= 3 	
	P: Vendor=03f0 ProdID=0857 Rev= 3.10 	
	S: Manufacturer=Qualcomm 	
	S: Product=Snapdragon(TM) X12 LTE-A 	
	S: SerialNumber=0123456789ABCDEF

	T: Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#= 3 Spd=480 MxCh= 0 	
	D: Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs= 1 	
	P: Vendor=1bc7 ProdID=1900 Rev= 3.10 	
	S: Manufacturer=Telit 	
	S: Product=Telit LN940 Mobile Broadband 	
	S: SerialNumber=0123456789ABCDEF
```

Serial port
```
/dev/ttyUSB1 & /dev/ttyUSB2
```
All band supported
4G: 1,2,3,4,5,7,8,12,13,17,18,19,20,21,25,26,28,29,30,38,39,40,41,66

For lock band slightly different from Dw5821e, this modem without "2" after lte

Lock B1
```
at^slband=lte,1
OK
at^slband?
^SLBAND: WCDMA,1,2,4,5,6,8,19
^SLBAND: LTE,1

OK
```
Lock B1 & B3
```
at^slband=lte,1,3
OK
at^slband?
^SLBAND: WCDMA,1,2,4,5,6,8,19
^SLBAND: LTE,1,3

OK
```
Reset to Auto
```
at^slband
OK
at^slband?
^SLBAND: WCDMA,1,2,4,5,6,8,19
^SLBAND: LTE,1,2,3,4,5,7,8,12,13,17,18,19,20,21,25,26,28,29,30,38,39,40,41,66

OK
```

lrdrdn, 2022-03-09:
# Huawei Me909s-821

vid & pid
```
12d1:15c1
12d1:1f1a
```
Serial port
```
/dev/ttyUSB0 & /dev/ttyUSB1
```
```
ati
Manufacturer: Huawei Technologies Co., Ltd.
Model: ME909s-821
Revision: 11.617.00.00.00
IMEI: xxx
+GCAP: +CGSM,+DS,+ES
```
Supported bands
4G: B1, B3, B8, B38, B40, B41

For lock band with bitmask
```
7FFFFFFFFFFFFFFF (CM_BAND_PREF_ANY) All bands
1 (CM_BAND_PREF_LTE_EUTRAN_BAND1) LTE BC1
2 (CM_BAND_PREF_LTE_EUTRAN_BAND2) LTE BC2
4 (CM_BAND_PREF_LTE_EUTRAN_BAND3) LTE BC3
8 (CM_BAND_PREF_LTE_EUTRAN_BAND4) LTE BC4
10 (CM_BAND_PREF_LTE_EUTRAN_BAND5) LTE BC5
20 (CM_BAND_PREF_LTE_EUTRAN_BAND6) LTE BC6
40 (CM_BAND_PREF_LTE_EUTRAN_BAND7) LTE BC7
80 (CM_BAND_PREF_LTE_EUTRAN_BAND8) LTE BC8
100 (CM_BAND_PREF_LTE_EUTRAN_BAND9) LTE BC9
200 (CM_BAND_PREF_LTE_EUTRAN_BAND10) LTE BC10
400 (CM_BAND_PREF_LTE_EUTRAN_BAND11) LTE BC11
800 (CM_BAND_PREF_LTE_EUTRAN_BAND12) LTE BC12
1000 (CM_BAND_PREF_LTE_EUTRAN_BAND13) LTE BC13
2000 (CM_BAND_PREF_LTE_EUTRAN_BAND14) LTE BC14
10000 (CM_BAND_PREF_LTE_EUTRAN_BAND17) LTE BC17
20000 (CM_BAND_PREF_LTE_EUTRAN_BAND18) LTE BC18
40000 (CM_BAND_PRE_LTE_EUTRAN_BAND19) LTE BC19
80000 (CM_BAND_PREF_LTE_EUTRAN_BAND20) LTE BC20
100000 (CM_BAND_PREF_LTE_EUTRAN_BAND21) LTE BC21
1000000 (CM_BAND_PREF_LTE_EUTRAN_BAND25) LTE BC25
2000000 (CM_BAND_PREF_LTE_EUTRAN_BAND26) LTE BC26
8000000 (CM_BAND_PREF_LTE_EUTRAN_BAND28) LTE BC28
100000000 (CM_BAND_PREF_LTE_EUTRAN_BAND33) LTE BC33
200000000 (CM_BAND_PREF_LTE_EUTRAN_BAND34) LTE BC34
400000000 (CM_BAND_PREF_LTE_EUTRAN_BAND35) LTE BC35
800000000 (CM_BAND_PREF_LTE_EUTRAN_BAND36) LTE BC36
1000000000 (CM_BAND_PREF_LTE_EUTRAN_BAND37) LTE BC37
2000000000 (CM_BAND_PREF_LTE_EUTRAN_BAND38) LTE BC38
4000000000 (CM_BAND_PREF_LTE_EUTRAN_BAND39) LTE BC39
8000000000 (CM_BAND_PREF_LTE_EUTRAN_BAND40) LTE BC40
10000000000 (CM_BAND_PREF_LTE_EUTRAN_BAND41) LTE BC41
20000000000 (CM_BAND_PREF_LTE_EUTRAN_BAND42) LTE BC42
40000000000 (CM_BAND_PREF_LTE_EUTRAN_BAND43) LTE BC43
40000000 (CM_BAND_PREF_NO_CHANGE) No band change
```

Lock 4G B1
```
AT^SYSCFGEX="03",3FFFFFFF,0,2,1,,
OK
AT^SYSCFGEX?
^SYSCFGEX: "03",3FFFFFFF,0,2,1

OK
```

Lock 4G B3
```
AT^SYSCFGEX="03",3FFFFFFF,0,2,4,,
OK
AT^SYSCFGEX?
^SYSCFGEX: "03",3FFFFFFF,0,2,4

OK
```
Lock 4G B1 & B3
```
AT^SYSCFGEX="03",3FFFFFFF,0,2,5,,
OK
AT^SYSCFGEX?
^SYSCFGEX: "03",3FFFFFFF,0,2,5

OK
```
All Band 4G 
```
AT^SYSCFGEX="03",3FFFFFFF,0,2,7FFFFFFFFFFFFFFF,,
OK
AT^SYSCFGEX?
^SYSCFGEX: "03",3FFFFFFF,0,2,7FFFFFFFFFFFFFFF

OK

```
Auto All band
```
AT^SYSCFGEX="00",3FFFFFFF,0,2,7FFFFFFFFFFFFFFF,,
OK
AT^SYSCFGEX?
^SYSCFGEX: "00",3FFFFFFF,0,2,7FFFFFFFFFFFFFFF

OK

```

4IceG, 2022-03-10:
OK, copy the files from the attachment to "/usr/share/modemband" and see if everything works.
[Huawei.zip](https://github.com/4IceG/luci-app-modemband/files/8220792/Huawei.zip)
[Foxconn.zip](https://github.com/4IceG/luci-app-modemband/files/8220793/Foxconn.zip)

lrdrdn, 2022-03-10:
I copied all files to /usr/share/modemband
```
03f0:0a57 (HP lt4220 MBIM)	OK
03f0:0857 (HP lt4220 QMI)	not detected
1bc7:1901 (Telit Ln940 MBIM)	OK
1bc7:1900 (Telit Ln940 QMI) 	not detected
```

03f0:0857 & 1bc7:1900, if I replace file from 0a57/1901 it will work

and LTE band there is B0 value detected like this
```
B0 B1 B2 B3 B4 B5 B7 B8 B12 B13 B17 B18 B19 B20 B21 B25 B26 B28 B29 B30 B38 B39 B40 B41 B66 
```

4IceG, 2022-03-10:
OK, copy the files and see if anything has improved.
[Foxconn2.zip](https://github.com/4IceG/luci-app-modemband/files/8221776/Foxconn2.zip)

Show me what the command returns:
modemband.sh getbands
modemband.sh getsupportedbands

lrdrdn, 2022-03-10:
```
root@ROOter:~# modemband.sh getbands
LTE 1 2 3 4 5 7 8 12 13 17 18 19 20 21 25 26 28 29 30 38 39 40 41 66
root@ROOter:~# modemband.sh getsupportedbands
1 2 3 4 5 7 8 12 13 17 18 19 20 21 25 26 28 29 30 38 39 40 41 66
```
everything works, but still displaying B0

[image]

For Huawei Me909s it's like before. 12d1:1f1a modem not detected

4IceG, 2022-03-10:
OK, copy the files and see if anything has improved
[Huawei2.zip](https://github.com/4IceG/luci-app-modemband/files/8222336/Huawei2.zip)
[Foxconn3.zip](https://github.com/4IceG/luci-app-modemband/files/8222345/Foxconn3.zip)
.

lrdrdn, 2022-03-10:
Huawei 12d1:15c1 use /dev/ttyUSB2 not USB1 

foxconn cannot getbands
```
root@ROOter:~#  modemband.sh getbands
tr: STRING2 cannot be empty

```
[image]

4IceG, 2022-03-10:
You yourself mentioned above that Huawei is dev/ttyUSB0 & /dev/ttyUSB1, but
ok, I assume that apart from the wrong port it works.

As for foxconn, tomorrow I will be over this session because the script
returns the data correctly, but then something else comes out for you.

Do this command to read the bands again and send me the result.

4IceG, 2022-03-10:
OK, copy the files and see if anything has improved.
[Foxconn3.zip](https://github.com/4IceG/luci-app-modemband/files/8227482/Foxconn3.zip)

Huawei Me909s-821 have these both /dev/ttyUSB2?
In the end, I didn't know if they were different or the same and made two versions.
12d1:15c1 ---> /dev/ttyUSB2
12d1:1f1a ---> /dev/ttyUSB1
or
12d1:15c1 ---> /dev/ttyUSB2
12d1:1f1a ---> /dev/ttyUSB2
[Huawei1xttyUSB2.zip](https://github.com/4IceG/luci-app-modemband/files/8227510/Huawei1xttyUSB2.zip)
[Huawei2xttyUSB2.zip](https://github.com/4IceG/luci-app-modemband/files/8227511/Huawei2xttyUSB2.zip)

closed 2022-03-11 by 4IceG

lrdrdn, 2022-03-11:
sorry I cannot test the modem now. may be tommorow or 2/3 days later. 
I think serial port on huawei determined by usb compossstion. It has 3 cfg if I plug in on openwrt detected cdc_ether. but if I plug in to rooter detected cdc_mbim

4IceG, 2022-03-11:
No problem, you check when you can.

There is an update to the modemband package, so do the update before
testing.

I know that the user can switch ports on the modem, so I added the port
selection options in the application. There are too many modems, it's hard
to make a super universal application.

lrdrdn, 2022-03-31:
Sorry for late repply.
Hp lt4220/t77w676 work perfectly.
For huawei me909s-821s I think default port should set to /dev/ttyUSB2 because USB1 only for rooter.

4IceG, 2022-03-31:
Nothing happened :) , I understand that @lrdrdn you downloaded the latest package versions (https://github.com/4IceG/luci-app-modemband/releases/latest) and tested.

I would not change the port settings, recently we also had a user on the forum who had a supported modem but he changed the ports in the modem and "scandal" because it does not work... There is an option to set the port, so "feel free to set it".

4IceG, 2022-03-31:
Because I don't remember if I added Foxconn T77W676

03f0:0a57 (HP lt4220 MBIM)
03f0:0857 (HP lt4220 QMI)
1bc7:1901 (Telit Ln940 MBIM)
1bc7:1900 (Telit Ln940 QMI)

I probably didn't add, I was waiting for confirmation that it works (update from Foxconn3.zip file).

4IceG, 2022-03-31:
Ok, so I sent the updates to Cezary. Now all we have to do is wait for the modemband package to be updated. After the upgrade, I will compile a new package.

4IceG, 2022-03-31:
@lrdrdn Can you check again what I sent you (this Foxconn3.zip)?

Cezary found some other commands for this Ln940 and we would like to confirm that this code what I posted works at 200% for these modems.

4IceG, 2022-04-01:
Hi,
I compiled a new modemband package but before publishing I wanted you @lrdrdn to test it to see if it works with all/any of your modems.
Thanks :)

[toTEST.zip](https://github.com/4IceG/luci-app-modemband/files/8398983/toTEST.zip)

lrdrdn, 2022-04-02:
Ln940 with [toTEST.zip](https://github.com/4IceG/luci-app-modemband/files/8398983/toTEST.zip) lock band & restore work fine, but B0 still visible

[image]

but if I replace with [Foxconn3.zip](https://github.com/4IceG/luci-app-modemband/files/8227482/Foxconn3.zip) like this

[image]

4IceG, 2022-04-03:
Hi, 
@lrdrdn can you show me the result of the command:
`sh -x /usr/bin/modemband.sh getbands`

The error occurs for a modem with what vid & pid?

lrdrdn, 2022-04-03:
1bc7:1900
1bc7:1901
03f0:0857
03f0:0a57

```
root@OpenWrt:~# sh -x /usr/bin/modemband.sh getbands
+ _DEVICE=
+ _DEFAULT_LTE_BANDS=
+ RES=/usr/share/modemband
+ awk '{gsub("="," ");
if ($0 ~ /Bus.*Lev.*Prnt.*Port.*/) {T=$0}
if ($0 ~ /Vendor.*ProdID/) {idvendor[T]=$3; idproduct[T]=$5}
if ($0 ~ /Product/) {product[T]=$3}}
END {for (idx in idvendor) {printf "%s%s\n%s%s%s\n", idvendor[idx], idproduct[idx], idvendor[idx], idproduct[idx], product[idx]}}' /sys/kernel/debug/usb/devices
+ _DEVS='0bda8153
0bda8153USB
1d6b0002
1d6b0002xHCI
1bc71900
1bc71900Telit
1d6b0002
1d6b0002EHCI
1d6b0003
1d6b0003xHCI
1d6b0001
1d6b0001Generic'
+ '[' -e /usr/share/modemband/0bda8153 ]
+ '[' -e /usr/share/modemband/0bda8153USB ]
+ '[' -e /usr/share/modemband/1d6b0002 ]
+ '[' -e /usr/share/modemband/1d6b0002xHCI ]
+ '[' -e /usr/share/modemband/1bc71900 ]
+ . /usr/share/modemband/1bc71900
+ _DEVICE=/dev/ttyUSB1
+ _DEFAULT_LTE_BANDS='1 2 3 4 5 7 8 12 13 17 18 19 20 21 25 26 28 29 30 38 39 40 41 66'
+ . /usr/share/modemband/_foxconn_common2
+ break
+ '[' -z /dev/ttyUSB1 ]
+ uci -q get 'modemband.@modemband[0].set_port'
+ _DEVICE1=
+ '[' -n  ]
+ '[' '!' -e /dev/ttyUSB1 ]
+ getbands
+ sms_tool -d /dev/ttyUSB1 at 'at^slband?'
+ O='
^SLBAND: WCDMA,1,2,4,5,6,8,19
^SLBAND: LTE,1,3,8,40
'
+ echo '
^SLBAND: WCDMA,1,2,4,5,6,8,19
^SLBAND: LTE,1,3,8,40
'
+ awk -F: '/LTE/{gsub("LTE,Enabled,","");gsub(","," ");print $2}'
+ xargs
LTE 1 3 8 40
+ exit 0
```

4IceG, 2022-04-03:
An update has come out, this version of the package should be working properly now :).

lrdrdn, 2022-04-04:
its broken

[image]

```
root@OpenWrt:~# modemband.sh
Modem: Telit LN940 Mobile Broadband
Supported LTE bands: 1 2 3 4 5 7 8 12 13 17 18 19 20 21 25 26 28 29 30 38 39 40 41 66
LTE bands: awk: cmd. line:1: Unexpected token

 1: FDD 2100 MHz
 2: FDD 1900 MHz
 3: FDD 1800 MHz
 4: FDD 1700 MHz
 5: FDD  850 MHz
 7: FDD 2600 MHz
 8: FDD  900 MHz
12: FDD  700 MHz
13: FDD  700 MHz
17: FDD  700 MHz
18: FDD  850 MHz
19: FDD  850 MHz
20: FDD  800 MHz
21: FDD 1500 MHz
25: FDD 1900 MHz
26: FDD  850 MHz
28: FDD  700 MHz
29: SDL  700 MHz
30: FDD 2300 MHz
38: TDD 2600 MHz
39: TDD 1900 MHz
40: TDD 2300 MHz
41: TDD 2500 MHz
66: FDD 1700 MHz
```

```
root@OpenWrt:~# sh -x /usr/bin/modemband.sh getbands
+ _DEVICE=
+ _DEFAULT_LTE_BANDS=
+ RES=/usr/share/modemband
+ awk '{gsub("="," ");
if ($0 ~ /Bus.*Lev.*Prnt.*Port.*/) {T=$0}
if ($0 ~ /Vendor.*ProdID/) {idvendor[T]=$3; idproduct[T]=$5}
if ($0 ~ /Product/) {product[T]=$3}}
END {for (idx in idvendor) {printf "%s%s\n%s%s%s\n", idvendor[idx], idproduct[idx], idvendor[idx], idproduct[idx], product[idx]}}' /sys/kernel/debug/usb/devices
+ _DEVS='0bda8153
0bda8153USB
1d6b0002
1d6b0002xHCI
1d6b0002
1d6b0002EHCI
1d6b0003
1d6b0003xHCI
1bc71900
1bc71900Telit
1d6b0001
1d6b0001Generic'
+ '[' -e /usr/share/modemband/0bda8153 ]
+ '[' -e /usr/share/modemband/0bda8153USB ]
+ '[' -e /usr/share/modemband/1d6b0002 ]
+ '[' -e /usr/share/modemband/1d6b0002xHCI ]
+ '[' -e /usr/share/modemband/1d6b0002 ]
+ '[' -e /usr/share/modemband/1d6b0002EHCI ]
+ '[' -e /usr/share/modemband/1d6b0003 ]
+ '[' -e /usr/share/modemband/1d6b0003xHCI ]
+ '[' -e /usr/share/modemband/1bc71900 ]
+ . /usr/share/modemband/1bc71900
+ _DEVICE=/dev/ttyUSB1
+ _DEFAULT_LTE_BANDS='1 2 3 4 5 7 8 12 13 17 18 19 20 21 25 26 28 29 30 38 39 40 41 66'
+ . /usr/share/modemband/_foxconn_common2
+ break
+ '[' -z /dev/ttyUSB1 ]
+ uci -q get 'modemband.@modemband[0].set_port'
+ _DEVICE1=
+ '[' -n  ]
+ '[' '!' -e /dev/ttyUSB1 ]
+ getbands
+ sms_tool -d /dev/ttyUSB1 at 'at^slband?'
+ O='
^SLBAND: WCDMA,1,2,4,5,6,8,19
^SLBAND: LTE,1,3,8,40
'
+ echo '
^SLBAND: WCDMA,1,2,4,5,6,8,19
^SLBAND: LTE,1,3,8,40
'
+ awk -F: '^\^SLBAND:.*LTE,/{gsub("LTE,","");gsub(","," ");print $2}'
+ xargs
awk: cmd. line:1: Unexpected token

+ exit 0
```

4IceG, 2022-04-04:
> + uci -q get ***@***.***[0].set_port'

Ok in file
/usr/share/modemband/_foxconn_common2
replace line

echo "$O" | awk -F: '^\^SLBAND:.*LTE,/{gsub("LTE,","");gsub(","," ");print
$2}' | xargs

to

echo "$O" | awk -F: '/^\^SLBAND:.*LTE,/{gsub("LTE,","");gsub(","," ");print
$2}' | xargs

lrdrdn, 2022-04-04:
ok its fix now. works fine

4IceG, 2022-04-04:
@lrdrdn I have compiled a new package with the fix, I hope that now everything will work properly for all modems.

4IceG, 2022-10-11:
I am sending a link to the file
https://www56.zippyshare.com/v/l9BuZI8M/file.html