4IceG/luci-app-modemband #3: Request add support modem

firkerword, 2022-03-08:
ZTE MF832s

firkerword, 2022-03-08:
AT+CGDCONT=1,"IP"
AT+CFUN=1
AT+CEREG=1
AT+CGREG?
AT+CEREG?
AT+ZGACT=1,1
AT+CGPADDR=1

firkerword, 2022-03-08:
I'm so sorry I only know so much.

firkerword, 2022-03-08:
https://drive.google.com/drive/folders/1wg6xgyQejmr39L_4xwv4ghT3-sJjajv4?usp=sharing

4IceG, 2022-03-08:
To add support for modem need:

   - exact name of modem and ew version number
   - VID and PID identifier on the USB bus
   - serial port used for communication with the modem ("diagnostic")
   - AT command to read the set bands together with an example result
   - an AT command to set specific bands
   - a list of all bands that can be set on the modem

If it is not there, we do not know, there is no chance of adding support.

closed 2022-03-08 by 4IceG