4IceG/luci-app-modemband #4: wsparcie dla modemów EM7565/EM7455/MC7455

przemekp1, 2022-03-08:
Cześć, 
może się przyda, ja korzystam z modelu EM7455
https://github.com/danielewood/sierra-wireless-modems, chyba, że chcesz abym wykonał jakieś komendy celem podania logów/parametrów tego modemu - to daj mi wytyczne.
Pozdrawiam

4IceG, 2022-03-08:
Siema,
Jak możesz to przygotuj wszystko wedle rozpiski dla tego EM7455:

- dokładna nazwa modemu i ew numer wersji
- identyfikator VID i PID na magistrali USB
- port szeregowy używany do komunikacji z modemem ("diagnostyczny")
- polecenie AT do odczytu ustawionych pasm wraz z przykładowym wynikiem
- polecenie AT do ustawienia określonych pasm
- listę wszystkich pasm możliwych do ustawienia w danym modemie

Nie ma to jak własna inicjatywa :)
Pozdrawiam

przemekp1, 2022-03-08:
może logi ze skrypu autoflash
cat: /dev/: Jest katalogiem
cat: /dev/: Jest katalogiem
cat: /dev/: Jest katalogiem
AT

OK
AT!ENTERCND="A710"

OK
AT!USBCOMP=1,1,0000100D

OK
AT!RESET

OK
AT

OK
ATE1

OK
AT!IMPREF?

!IMPREF: 
 preferred fw version:    02.24.05.06
 preferred carrier name:  ORANGE-EU
 preferred config name:   ORANGE-EU_002.022_000
 current fw version:      02.24.05.06
 current carrier name:    ORANGE-EU
 current config name:     ORANGE-EU_002.022_000

OK
AT!GOBIIMPREF?

!GOBIIMPREF: 
 preferred fw version:    02.24.05.06
 preferred carrier name:  ORANGE-EU
 preferred config name:   ORANGE-EU_002.022_000
 current fw version:      02.24.05.06
 current carrier name:    ORANGE-EU
 current config name:     ORANGE-EU_002.022_000

OK
AT!USBSPEED?

ERROR
AT!USBSPEED=?

ERROR
AT!USBCOMP?

ERROR
AT!USBCOMP=?

ERROR
AT!USBVID?

ERROR
AT!USBPID?

ERROR
AT!USBPRODUCT?

ERROR
AT!PRIID?

PRI Part Number: 9904336
Revision: 001.007
Customer: Dell
[... 393 lines cut ...]
PRI  FF   GOOD   0   0 0      002.022_000 02.24.05.06_ORANGE-EU
PRI  FF   GOOD   0   0 0      001.023_000 02.33.03.00_SIERRA
Max PRI images: 50

OK
AT

OK
AT!IMAGE=0

OK
AT!IMAGE?

TYPE SLOT STATUS LRU FAILURES UNIQUE_ID   BUILD_ID
FW   1    EMPTY  0   0 0                  
FW   2    EMPTY  0   0 0                  
FW   3    EMPTY  0   0 0                  
FW   4    EMPTY  0   0 0                  
Max FW images: 4
Active FW image is at slot 255

TYPE SLOT STATUS LRU FAILURES UNIQUE_ID   BUILD_ID
Max PRI images: 50

OK

przemekp1, 2022-03-08:
i jeszcze logi z OPenwrt - kernel.log
[    0.000000] Linux version 5.4.179 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16495-bf0c965af0)) #0 Wed Feb 16 20:29:10 2022
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
[    0.000000] MIPS: machine is Zbtlink ZBT-WE826 (16M)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00027e90
[    0.000000] Readback ErrCtl register=00027e90
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 122408K/131072K available (5070K kernel code, 206K rwdata, 632K rodata, 1188K init, 213K bss, 8664K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from 0x805c7a28 with crng_init=0
[    0.000000] CPU Clock: 580MHz
[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
[    0.000000] systick: enable autosleep mode
[    0.000000] systick: running - mult: 214748, shift: 32
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
[    0.000011] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
[    0.007622] Calibrating delay loop... 385.02 BogoMIPS (lpj=770048)
[    0.045503] pid_max: default: 32768 minimum: 301
[    0.050197] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.057256] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.071687] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.081162] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.087852] pinctrl core: initialized pinctrl subsystem
[    0.094482] NET: Registered protocol family 16
[    0.579713] mt7620-pci 10140000.pcie: PCIE0 no card, disable it(RST&CLK)
[    0.586215] mt7620-pci: probe of 10140000.pcie failed with error -1
[    0.615032] rt2880_gpio 10000600.gpio: registering 24 gpios
[    0.620516] rt2880_gpio 10000600.gpio: registering 24 irq handlers
[    0.626874] rt2880_gpio 10000638.gpio: registering 16 gpios
[    0.632308] rt2880_gpio 10000638.gpio: registering 16 irq handlers
[    0.638666] rt2880_gpio 10000688.gpio: registering 1 gpios
[    0.644016] rt2880_gpio 10000688.gpio: registering 1 irq handlers
[    0.656056] clocksource: Switched to clocksource systick
[    0.662399] thermal_sys: Registered thermal governor 'step_wise'
[    0.662861] NET: Registered protocol family 2
[    0.673074] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.681010] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.689166] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.696583] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.703416] TCP: Hash tables configured (established 1024 bind 1024)
[    0.709732] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.716094] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.723131] NET: Registered protocol family 1
[    0.727369] PCI: CLS 0 bytes, default 32
[    0.734326] rt-timer 10000100.timer: maximum frequency is 1220Hz
[... 120 lines cut ...]
[   47.199741] br-lan: port 1(eth0.1) entered blocking state
[   47.205330] br-lan: port 1(eth0.1) entered disabled state
[   47.211314] device eth0.1 entered promiscuous mode
[   47.216244] device eth0 entered promiscuous mode
[   47.249825] br-lan: port 1(eth0.1) entered blocking state
[   47.255418] br-lan: port 1(eth0.1) entered forwarding state
[   48.179785] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   58.746463] ieee80211 phy0: rt2800_rf_self_txdc_cal: Info - RF Tx self calibration start
[   58.755306] ieee80211 phy0: rt2800_rf_self_txdc_cal: Info - RF Tx self calibration end
[   65.497436] ieee80211 phy0: rt2800_loft_iq_calibration: Info - LOFT Calibration Done!
[   66.010650] ieee80211 phy0: rt2800_iq_search: Info - IQCalibration Start!
[... cut ...]
[   68.112575] ieee80211 phy0: rt2800_rxiq_calibration: Info - RXIQ G_imb=0, Ph_rx=2
[   68.275039] br-lan: port 2(wlan0) entered blocking state
[   68.280557] br-lan: port 2(wlan0) entered disabled state
[   68.286463] device wlan0 entered promiscuous mode
[   69.055617] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   69.062417] br-lan: port 2(wlan0) entered blocking state
[   69.067903] br-lan: port 2(wlan0) entered forwarding state
[   83.172249] 8021q: adding VLAN 0 to HW filter on device wwan0

przemekp1, 2022-03-08:
i system log
ue Mar  8 18:39:45 2022 kern.info kernel: [    0.081162] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.087852] pinctrl core: initialized pinctrl subsystem
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.094482] NET: Registered protocol family 16
Tue Mar  8 18:39:45 2022 kern.err kernel: [    0.579713] mt7620-pci 10140000.pcie: PCIE0 no card, disable it(RST&CLK)
Tue Mar  8 18:39:45 2022 kern.warn kernel: [    0.586215] mt7620-pci: probe of 10140000.pcie failed with error -1
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.615032] rt2880_gpio 10000600.gpio: registering 24 gpios
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.620516] rt2880_gpio 10000600.gpio: registering 24 irq handlers
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.626874] rt2880_gpio 10000638.gpio: registering 16 gpios
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.632308] rt2880_gpio 10000638.gpio: registering 16 irq handlers
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.638666] rt2880_gpio 10000688.gpio: registering 1 gpios
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.644016] rt2880_gpio 10000688.gpio: registering 1 irq handlers
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.656056] clocksource: Switched to clocksource systick
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.662399] thermal_sys: Registered thermal governor 'step_wise'
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.662861] NET: Registered protocol family 2
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.673074] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.681010] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.689166] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.696583] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.703416] TCP: Hash tables configured (established 1024 bind 1024)
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.709732] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.716094] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.723131] NET: Registered protocol family 1
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.727369] PCI: CLS 0 bytes, default 32
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.734326] rt-timer 10000100.timer: maximum frequency is 1220Hz
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.742528] workingset: timestamp_bits=14 max_order=15 bucket_order=1
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.757517] squashfs: version 4.0 (2009/01/31) Phillip Lougher
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.763179] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.788581] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 252)
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.798893] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.808218] printk: console [ttyS0] disabled
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.812393] 10000c00.uartlite: ttyS0 at MMIO 0x10000c00 (irq = 20, base_baud = 2500000) is a Palmchip BK-3103
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.822015] printk: console [ttyS0] enabled
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.830479] printk: bootconsole [early0] disabled
Tue Mar  8 18:39:45 2022 kern.warn kernel: [    0.849597] spi spi0.0: force spi mode3
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.863936] spi-nor spi0.0: w25q128 (16384 Kbytes)
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.868911] 4 fixed-partitions partitions found on MTD device spi0.0
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.875405] Creating 4 MTD partitions on "spi0.0":
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.880320] 0x000000000000-0x000000030000 : "u-boot"
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.886582] 0x000000030000-0x000000040000 : "u-boot-env"
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.893222] 0x000000040000-0x000000050000 : "factory"
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.899588] 0x000000050000-0x000001000000 : "firmware"
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.910051] 2 uimage-fw partitions found on MTD device firmware
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.916167] Creating 2 MTD partitions on "firmware":
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.921263] 0x000000000000-0x0000001e6aad : "kernel"
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.927552] 0x0000001e6aad-0x000000fb0000 : "rootfs"
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.933765] mtd: device 5 (rootfs) set to be root filesystem
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.941865] 1 squashfs-split partitions found on MTD device rootfs
Tue Mar  8 18:39:45 2022 kern.notice kernel: [    0.948268] 0x000000525000-0x000000fb0000 : "rootfs_data"
Tue Mar  8 18:39:45 2022 kern.info kernel: [    0.965885] gsw: setting port4 to ephy mode
[... cut ...]
Tue Mar  8 18:42:52 2022 daemon.info dnsmasq[3059]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Tue Mar  8 18:42:52 2022 daemon.info dnsmasq-dhcp[3059]: read /etc/ethers - 0 addresses
Tue Mar  8 18:42:53 2022 daemon.info dnsmasq[3059]: read /etc/hosts - 4 addresses
Tue Mar  8 18:42:53 2022 daemon.info dnsmasq[3059]: read /tmp/hosts/odhcpd - 1 addresses
Tue Mar  8 18:42:53 2022 daemon.info dnsmasq[3059]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Tue Mar  8 18:42:53 2022 daemon.info dnsmasq-dhcp[3059]: read /etc/ethers - 0 addresses

4IceG, 2022-03-08:
Napisałem konkretnie czego potrzebuje, specjalnie abym nie musiał wertować tysiąca stron. 
Czyli co.. mam sobie sam poszukać potrzebnych danych?

przemekp1, 2022-03-09:
niestety nie jestem tutaj asem w tej dziedzinie
podaj jakieś komendy jak to zrobić - to zbiorę Ci to do logicznej kupy

4IceG, 2022-03-09:
Przemek, ja tez nie znam polecen dla tego modemu i nie sa mi one do zycia
potrzebne. A mowiac ze sie nie znasz idziesz na latwizne. Tematow z tym
modemem jest w sieci od groma, a ja podalem konkretnie co mi jest potrzebne.

Podstawa podstaw to wynik polecenia:
"cat /sys/kernel/debug/usb/devices"

Do tego musze znac port do komunikacji z modemem, gdzies mi sie rzucilo w
oczy ze ktos z nim gadal przez "/dev/ttyUSB4" to tez do sprawdzenia.

closed 2022-03-11 by 4IceG