4IceG/luci-app-modemband #9: Support for Sierra Wireless EM7565

PR-NCE, 2023-04-21:
exact name of modem and ew version number:
Sierra Wireless EM7565

VID and PID identifier on the USB bus:
```
AT!USBPID?
!USBPID:
APP : 9091
BOOT: 9090

AT!USBVID?
!USBVID: 1199
```
serial port used for communication with the modem ("diagnostic"):
```/dev/ttyUSB2```

a list of all bands that can be set on the modem:
AT command to read the set bands together with an example result:
```
AT!BAND=?
Index, Name,                        GW Band Mask     L Band Mask 1    L Band Mask 2    TDS Band Mask    L Band Mask 3    L Band Mask 4
00, All bands                     100600000FC00000 00000100BA0E19DF 0000000000000002 0000000000000000 0000000000000000 0000000000000000
01, Europe 3G                     0002000000400000 0000000000000000 0000000000000000 0000000000000000 0000000000000000 0000000000000000
02, North America 3G              0000000004800000 0000000000000000 0000000000000000 0000000000000000 0000000000000000 0000000000000000
06, Europe                        0002000000400000 00000000000801C5 0000000000000000 0000000000000000 0000000000000000 0000000000000000
07, North America                 0000000004800000 000001000200185A 0000000000000002 0000000000000000 0000000000000000 0000000000000000
08, WCDMA ALL                     100600000FC00000 0000000000000000 0000000000000000 0000000000000000 0000000000000000 0000000000000000
09, LTE ALL                       0000000000000000 00000100BA0E19DF 0000000000000002 0000000000000000 0000000000000000 0000000000000000

                                                                    0000000000000002 - B66
                                                   0000010000000000 - B41
                                                   0000000080000000 - B32
                                                   0000000020000000 - B30
                                                   0000000010000000 - B29
                                                   0000000008000000 - B28
                                                   0000000002000000 - B26
                                                   0000000000080000 - B20
                                                   0000000000040000 - B19
                                                   0000000000020000 - B18
                                                   0000000000001000 - B13
                                                   0000000000000800 - B12
                                                   0000000000000100 - B9
                                                   0000000000000080 - B8
                                                   0000000000000040 - B7
                                                   0000000000000010 - B5
                                                   0000000000000008 - B4
                                                   0000000000000004 - B3
                                                   0000000000000002 - B2
                                                   0000000000000001 - B1
                                  1000000000000000 - B19 (850)
                                  0004000000000000 - B9 (1700)
                                  0002000000000000 - B8  (900)
                                  0000000008000000 - B6  (800)
                                  0000000004000000 - B5  (850)
                                  0000000002000000 - B4 (1700)
                                  0000000001000000 - B3 (1700)
                                  0000000000800000 - B2 (1900)
                                  0000000000400000 - B1 (2100)
```
an AT command to set specific bands:

```
AT!ENTERCND="A710"
```
add new band groups:
```
AT!BAND=10,"Only B2",0,0000000000000002 
AT!BAND=11,"Only B4",0,000000000000008
AT!BAND=0F,"B2 and B4",0,000000000000000A
```
set to only B2 and B4 by selecting group 0F:
```
AT!BAND=0F

AT!RESET
```
[Link to complete and clear explanation of how to set specific bands on almost all of popular Sierra Wireless modems including EM7565](https://wirelessjoint.com/viewtopic.php?f=16&t=364&p=2034&hilit=band+locking+spreadsheet#p2034)

Please let me know if I missed anything that you might need, thank you in advance.

4IceG, 2023-04-23:
Hi @PR-NCE , there is a big problem with adding support for Sierra Wireless modems. Through the band grouping mechanism, I can't easily read and set the bands.

PR-NCE, 2023-05-04:
Hi @4IceG, are you open to the idea of adding support for sierra modems if I or someone else write the code for it? and can you please point me to the file in your repository that should be modified (to save me time as i haven't looked to see how you programmed this luci app) e.g. the exact part that is responsible to handle band looking for Quectel modems.

4IceG, 2023-05-04:
Hi @PR-NCE , I'm open to code. If you add simple code that is compatible with the modemband package.
https://github.com/4IceG/luci-app-modemband/tree/main/modemband

My package for LuCI is unimportant, all operations are done by modemband package, I only made the gui for the user.
For example, for Quectel EM12-G
https://github.com/4IceG/luci-app-modemband/blob/main/modemband/files/usr/share/modemband/2c7c0512

obsy, 2023-12-16:
Hi,
could you again show output of:
```
AT!ENTERCND="A710"
AT!BAND=?
AT!BAND?

```