4IceG/luci-app-modemband #10: Add Carrier Aggregation

gold502, 2023-06-18:
Hello, @4IceG, ZTE MF289F is a router which support carrier aggregation (it can make 3 LTE band aggregation at the same time) , Could you add the function on the router? I suggest that luci-app-modemband can modify carrier aggregation manually, Then users can get better experience.

4IceG, 2023-06-18:
Hi @gold502 
I don't know what additional function would be added.

After installing the package, user can choose the preferred LTE bands. User changes to bands are visible from LuCI.
Package works, it is easy to use, there is nothing to add here.

gold502, 2023-06-19:
Hello, @4IceG
Carrier aggregation is a technology which can use multiple LTE band at the same time, for example, if I use B3+B8+B41 at the same time, then I will get better network speed, but It seems that I can only use one band in luci-app-modemband, it can't make carrier aggregation.

4IceG, 2023-06-19:
Hi @gold502 ,
In that case, you started the topic wrong, you should write right away that there is some error in the selection of bands.

Run the commands in the console:
`modemband.sh`
and
`sms_tool -d /dev/ttyUSB1 at "at+znlockband?"`
`sms_tool -d /dev/ttyUSB1 at "at+zcainfo?"`
and show their result.

It's strange because no one has reported bugs before.

gold502, 2023-06-20:
Hello @4IceG ,
Sorry, I made a mistake, I have already run the commands, and the carrier aggregation was worked successfully, The reason why the network speed has not changed is that the plug is broken and the power supply is insufficient, resulting in the inability to fully utilize the performance of the router.Thank you very much!

closed 2023-06-20 by gold502