4IceG/luci-app-modemband #11: hello i am new and i using openwrt need modemband

khalid64392, 2023-07-03:
my router model
tl-mr6400 v5
can you help me?

4IceG, 2023-07-03:
Hi @khalid64392 , much more information is needed to add modem support

- exact name of modem and ew version number
- VID and PID identifier on the USB bus
- serial port used for communication with the modem ("diagnostic")
- AT command to read the set bands together with an example result
 - an AT command to set specific bands
 - a list of all bands that can be set on the modem

You can check if this package [luci-app-mmconfig](https://github.com/koshev-msk/modemfeed/tree/master/luci/applications/luci-app-mmconfig) supports your modem.

khalid64392, 2023-07-03:
can i install luci-app-mmconfig with ssh package?

4IceG, 2023-07-04:
You can add this repository `https://openwrt.132lan.ru/packages/21.02/packages/` to your router or download the file [luci-app-mmconfig_0.0.5-3_all.ipk](https://openwrt.132lan.ru/packages/21.02/packages/mipsel_24kc/modemfeed/luci-app-mmconfig_0.0.5-3_all.ipk) manually to install the package via ssh.

khalid64392, 2023-07-06:
when install  it say
Connection error: Invalid SSL certificate

4IceG, 2023-07-06:
Hi @khalid64392 
Did you install the way to add the repository?

Instructions on how to add a repository -> https://github.com/4IceG/Modem-extras, you only replace the links.

```
src/gz repo https://openwrt.132lan.ru/packages/21.02/packages/

wget https://openwrt.132lan.ru/packages/21.02/packages/repo.pub -O /tmp/repo.pub
opkg-key add /tmp/repo.pub

```

4IceG, 2023-07-06:
You can also script whatever you like.
If the repo doesn't work and you can't install the package manually (by downloading it), report the problem to the creator of the package.

khalid64392, 2023-07-14:
Help?