4IceG/luci-app-modemband #12: Wsparcie dla EM7455 DW5811e Snapdragon™ X7 LTE

przemekp1, 2023-08-02:
Cześć,
czy będzie wsparcie dla tego modelu?
[image]

4IceG, 2023-08-03:
Cześć,
Ciężko powiedzieć czy będzie wsparcie. Modemy z rodziny Sierra Wireless mają dość specyficzny sposób ustawiania/odczytu pasm. Widziałem kalkulatory w Excelu, które to wszystko wyliczały, ale już nie pamiętam czy uniwersalnie, czy według przyjętego schematu. Na tą chwilę jest brak pomysłu jak to opisać.

przemekp1, 2023-08-05:
przyjmuje komendy
 sms_tool -d /dev/ttyUSB2 at "AT!BAND=?"
AT!BAND=?
Index, Name
00, All bands
01, Europe 3G
02, North America 3G
06, Europe
07, North America
08, WCDMA ALL
09, LTE ALL

sms_tool -d /dev/ttyUSB2 at "AT!ENTERCND=”A710”"

sms_tool -d /dev/ttyUSB2 at "AT!BAND=6"

https://github.com/danielewood/sierra-wireless-modems
tutaj są takie propozycje
np. AT!BAND=10,"B2B1-B3B7B20",0000000000C00000,0000000000080044

4IceG, 2023-08-06:
Siema
Widzę @przemekp1 , że ostro walczysz z tematem.

Polecenia są znane, problem jest z prostym schematem odczytu/zapisu pasm i przeniesieniem tego na postać kodu.

Wystarczy spojrzeć na to co się dzieje w tym temacie https://wirelessjoint.com/viewtopic.php?t=364 lub np. tu https://wirelessjoint.com/viewtopic.php?t=1034.

Kalkulator w Excelu
[Create Custom Bands (11-02-19).xlsx](https://github.com/4IceG/luci-app-modemband/files/12268857/Create.Custom.Bands.11-02-19.xlsx)

przemekp1, 2023-12-08:
sms_tool -d /dev/ttyUSB2 at "AT!ENTERCND="A710""
AT!ENTERCND=A710
sms_tool -d /dev/ttyUSB2 at "AT!BAND=?"
AT!BAND=?
Index, Name
00, All bands
01, Europe 3G
02, North America 3G
06, Europe
07, North America
08, WCDMA ALL
09, LTE ALL

4IceG, 2023-12-08:
@przemekp1 Dostałem informacje zwrotną, że nie taka powinna być odpowiedź, teraz sprawdziłem i o ile się nie pomyliłem to zapis polecenia dla sms_tool powinien wyglądać tak:
`sms_tool -d /dev/ttyUSB2 at "AT!ENTERCND=\"A710\""`

obsy, 2023-12-15:
@przemekp1 pokaż wynik
```
cat /sys/kernel/debug/usb/devices
```

przemekp1, 2023-12-15:
cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.15
S:  Manufacturer=Linux 5.15.137 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.15
S:  Manufacturer=Linux 5.15.137 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs=  1
P:  Vendor=413c ProdID=81b6 Rev= 0.06
S:  Manufacturer=Sierra Wireless, Incorporated
S:  Product=DW5811e Snapdragon™ X7 LTE
S:  SerialNumber=LF61544850011013
C:* #Ifs= 5 Cfg#= 1 Atr=a0 MxPwr=504mA
A:  FirstIf#=12 IfCount= 2 Cls=02(comm.) Sub=0e Prot=00
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=ff Driver=qcserial
E:  Ad=81(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=qcserial
E:  Ad=83(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 3 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=qcserial
E:  Ad=85(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=84(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#=12 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=0e Prot=00 Driver=cdc_mbim
E:  Ad=87(I) Atr=03(Int.) MxPS=  64 Ivl=32ms
I:  If#=13 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
I:* If#=13 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
E:  Ad=86(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms

obsy, 2023-12-15:
Zainstaluj na routerze modemband jak nie masz, zrób z konsoli następnie
```
wget -O /usr/share/modemband/413c81b6 http://dl.eko.one.pl/test/413c81b6
```
Powinieneś mieć możliwość  odczytu i ustawienia pasm. W kodzie jest sporo uproszczeń i na dodatek po ustawieniu pasm zrobiony jest reset modem, ale powinno działać. Masz tak samo jak ja Della DW5811e.

przemekp1, 2023-12-15:
[image]

przemekp1, 2023-12-15:
[image]

przemekp1, 2023-12-15:
[image]

obsy, 2023-12-15:
zrób 
```
modemband json
```

I pokaż co wysszło.

przemekp1, 2023-12-15:
# modemband json
-ash: modemband: not found

obsy, 2023-12-15:
```
modemband.sh  json
```

oczywiście...

przemekp1, 2023-12-15:
modemband.sh  json
{ "modem": "Sierra Wireless EM7455\/MC7455\/DW5811e", "supported": [ { "band": 1, "txt": "FDD 2100 MHz" }, { "band": 2, "txt": "FDD 1900 MHz" }, { "band": 3, "txt": "FDD 1800 MHz" }, { "band": 4, "txt": "FDD 1700 MHz" }, { "band": 5, "txt": "FDD  850 MHz" }, { "band": 7, "txt": "FDD 2600 MHz" }, { "band": 8, "txt": "FDD  900 MHz" }, { "band": 12, "txt": "FDD  700 MHz" }, { "band": 13, "txt": "FDD  700 MHz" }, { "band": 20, "txt": "FDD  800 MHz" }, { "band": 25, "txt": "FDD 1900 MHz" }, { "band": 26, "txt": "FDD  850 MHz" }, { "band": 29, "txt": "SDL  700 MHz" }, { "band": 41, "txt": "TDD 2500 MHz" } ], "enabled": [ ] }

4IceG, 2023-12-15:
Aby odczarować komunikat w niebieskim polu, klucz "enabled" nie może być pusty.

przemekp1, 2023-12-15:
uparte to ustrojstwo ¯\_(ツ)_/¯

obsy, 2023-12-15:
Nadal jest źle bo enabled nic nie zwraca. 
```
sh -x /usr/bin/modemband.sh getbands
```

pokaż.

przemekp1, 2023-12-15:
sh -x /usr/bin/modemband.sh getbands
+ _DEVICE=
+ _DEFAULT_LTE_BANDS=
+ _DEFAULT_5GNSA_BANDS=
+ _DEFAULT_5GSA_BANDS=
+ RES=/usr/share/modemband
+ awk '{gsub("="," ");
if ($0 ~ /Bus.*Lev.*Prnt.*Port.*/) {T=$0}
if ($0 ~ /Vendor.*ProdID/) {idvendor[T]=$3; idproduct[T]=$5}
if ($0 ~ /Product/) {product[T]=$3}}
END {for (idx in idvendor) {printf "%s%s\n%s%s%s\n", idvendor[idx], idproduct[idx], idvendor[idx], idproduct[idx], pr                                        oduct[idx]}}' /sys/kernel/debug/usb/devices
+ _DEVS='413c81b6
413c81b6DW5811e
1d6b0002
1d6b0002xHCI
1d6b0003
1d6b0003xHCI'
+ '[' -e /usr/share/modemband/413c81b6 ]
+ . /usr/share/modemband/413c81b6
+ _DEVICE=/dev/ttyUSB2
+ break
+ '[' -z /dev/ttyUSB2 ]
+ uci -q get 'modemband.@modemband[0].set_port'
+ _DEVICE1=/dev/ttyUSB2
+ '[' -n /dev/ttyUSB2 ]
+ _DEVICE=/dev/ttyUSB2
+ '[' '!' -e /dev/ttyUSB2 ]
+ getbands
+ sms_tool -d /dev/ttyUSB2 at 'AT!ENTERCND="A710"'
+ sms_tool -d /dev/ttyUSB2 at 'AT!BAND?'
+ O='AT!BAND?
Index, Name,                        GW Band Mask     L Band Mask      TDS Band Mask
06, Europe,                         0002000000400000 00000000000800C5 0000000000000000

'
+ echo 'AT!BAND?
Index, Name,                        GW Band Mask     L Band Mask      TDS Band Mask
06, Europe,                         0002000000400000 00000000000800C5 0000000000000000

'
+ awk '/^[0-9a-fA-F][0-9a-fA-F],\s/{printf "0x%s\n", $5}'
+ HEX=0x0000000000000000
+ '[' 0x0000000000000000 '=' 0x ]
+ hextobands 0x0000000000000000
+ BANDS=
+ HEX=0x0000000000000000
+ LEN=18
+ '[' 18 -gt 18 ]
+ seq 0 63
+ POW=1
+ T=0
+ '[' x0 '=' x1 ]
+ POW=2
+ T=0
+ '[' x0 '=' x2 ]
+ POW=4
+ T=0
+ '[' x0 '=' x4 ]
+ POW=8
+ T=0
+ '[' x0 '=' x8 ]
+ POW=16
+ T=0
+ '[' x0 '=' x16 ]
+ POW=32
+ T=0
+ '[' x0 '=' x32 ]
+ POW=64
+ T=0
+ '[' x0 '=' x64 ]
+ POW=128
+ T=0
+ '[' x0 '=' x128 ]
+ POW=256
+ T=0
+ '[' x0 '=' x256 ]
+ POW=512
+ T=0
+ '[' x0 '=' x512 ]
+ POW=1024
+ T=0
+ '[' x0 '=' x1024 ]
+ POW=2048
+ T=0
+ '[' x0 '=' x2048 ]
+ POW=4096
+ T=0
+ '[' x0 '=' x4096 ]
+ POW=8192
+ T=0
+ '[' x0 '=' x8192 ]
+ POW=16384
+ T=0
+ '[' x0 '=' x16384 ]
+ POW=32768
+ T=0
+ '[' x0 '=' x32768 ]
+ POW=65536
+ T=0
+ '[' x0 '=' x65536 ]
+ POW=131072
+ T=0
+ '[' x0 '=' x131072 ]
+ POW=262144
+ T=0
+ '[' x0 '=' x262144 ]
+ POW=524288
+ T=0
+ '[' x0 '=' x524288 ]
+ POW=1048576
+ T=0
+ '[' x0 '=' x1048576 ]
+ POW=2097152
+ T=0
+ '[' x0 '=' x2097152 ]
+ POW=4194304
+ T=0
+ '[' x0 '=' x4194304 ]
+ POW=8388608
+ T=0
+ '[' x0 '=' x8388608 ]
+ POW=16777216
+ T=0
+ '[' x0 '=' x16777216 ]
+ POW=33554432
+ T=0
+ '[' x0 '=' x33554432 ]
+ POW=67108864
+ T=0
+ '[' x0 '=' x67108864 ]
+ POW=134217728
+ T=0
+ '[' x0 '=' x134217728 ]
+ POW=268435456
+ T=0
+ '[' x0 '=' x268435456 ]
+ POW=536870912
+ T=0
+ '[' x0 '=' x536870912 ]
+ POW=1073741824
+ T=0
+ '[' x0 '=' x1073741824 ]
+ POW=2147483648
+ T=0
+ '[' x0 '=' x2147483648 ]
+ POW=4294967296
+ T=0
+ '[' x0 '=' x4294967296 ]
+ POW=8589934592
+ T=0
+ '[' x0 '=' x8589934592 ]
+ POW=17179869184
+ T=0
+ '[' x0 '=' x17179869184 ]
+ POW=34359738368
+ T=0
+ '[' x0 '=' x34359738368 ]
+ POW=68719476736
+ T=0
+ '[' x0 '=' x68719476736 ]
+ POW=137438953472
+ T=0
+ '[' x0 '=' x137438953472 ]
+ POW=274877906944
+ T=0
+ '[' x0 '=' x274877906944 ]
+ POW=549755813888
+ T=0
+ '[' x0 '=' x549755813888 ]
+ POW=1099511627776
+ T=0
+ '[' x0 '=' x1099511627776 ]
+ POW=2199023255552
+ T=0
+ '[' x0 '=' x2199023255552 ]
+ POW=4398046511104
+ T=0
+ '[' x0 '=' x4398046511104 ]
+ POW=8796093022208
+ T=0
+ '[' x0 '=' x8796093022208 ]
+ POW=17592186044416
+ T=0
+ '[' x0 '=' x17592186044416 ]
+ POW=35184372088832
+ T=0
+ '[' x0 '=' x35184372088832 ]
+ POW=70368744177664
+ T=0
+ '[' x0 '=' x70368744177664 ]
+ POW=140737488355328
+ T=0
+ '[' x0 '=' x140737488355328 ]
+ POW=281474976710656
+ T=0
+ '[' x0 '=' x281474976710656 ]
+ POW=562949953421312
+ T=0
+ '[' x0 '=' x562949953421312 ]
+ POW=1125899906842624
+ T=0
+ '[' x0 '=' x1125899906842624 ]
+ POW=2251799813685248
+ T=0
+ '[' x0 '=' x2251799813685248 ]
+ POW=4503599627370496
+ T=0
+ '[' x0 '=' x4503599627370496 ]
+ POW=9007199254740992
+ T=0
+ '[' x0 '=' x9007199254740992 ]
+ POW=18014398509481984
+ T=0
+ '[' x0 '=' x18014398509481984 ]
+ POW=36028797018963968
+ T=0
+ '[' x0 '=' x36028797018963968 ]
+ POW=72057594037927936
+ T=0
+ '[' x0 '=' x72057594037927936 ]
+ POW=144115188075855872
+ T=0
+ '[' x0 '=' x144115188075855872 ]
+ POW=288230376151711744
+ T=0
+ '[' x0 '=' x288230376151711744 ]
+ POW=576460752303423488
+ T=0
+ '[' x0 '=' x576460752303423488 ]
+ POW=1152921504606846976
+ T=0
+ '[' x0 '=' x1152921504606846976 ]
+ POW=2305843009213693952
+ T=0
+ '[' x0 '=' x2305843009213693952 ]
+ POW=4611686018427387904
+ T=0
+ '[' x0 '=' x4611686018427387904 ]
+ POW=-9223372036854775808
+ T=0
+ '[' x0 '=' x-9223372036854775808 ]
+ '[' -n  ]
+ echo

+ exit 0

obsy, 2023-12-15:
Moja wina, poczekaj chwilę.

obsy, 2023-12-15:
Pobierz jeszcze raz i sprawdź.

przemekp1, 2023-12-15:
chyba poszło
[image]
[image]

obsy, 2023-12-15:
I faktycznie ustawiłeś sobie tylko B20?

przemekp1, 2023-12-15:
tak i nie chce już innych zmienić ustawień 
i nie ma neta

obsy, 2023-12-15:
Nikt nie powiedział że zmianie bandu będziesz miał internet. Ustaw inne pasmo.

Można zamykać. Ograniczenia na dziś:
- nie parsuję masek bitowych, jak się trafi inny model z innymi maskami to przestanie działać
-  wykorzystuję slot "10" do customowego bandu, nie parsuje i nie szukam wolnego
-  nie interesuje mnie także to czy coś tam już było czy nie
- zakładam że slot 00 nazywa się "All bands" i zawiera wszystkie dostępne bandy. 
- 
- Później się inaczej przeparsuje żeby to jakoś było bardziej uniwersalnie.

przemekp1, 2023-12-15:
NET WRÓCIŁ
zmieniłem na 3 pasma, które obługuje mój nadajnik
[image]
ale może to zasięg jest słaby i trzyma na B20?
[image]

4IceG, 2023-12-15:
@przemekp1 sprawdz sobie poleceniem AT!LTECA? dostepne kombinacje agregacji. Nie wiem na ile to prawda ale powiedziano mi że nie wskazane jest mieszanie pasm, wybór pasm z roznych wariantow (kombinacji).

przemekp1, 2023-12-15:
sms_tool -d /dev/ttyUSB2 at "AT!LTECA?"
AT!LTECA?
Hardware:
LTEB1: B8,
LTEB2: B2, B5, B12, B13, B29,
LTEB3: B7, B20,
LTEB4: B4, B5, B12, B13, B29,
LTEB5: B2, B4, B30,
LTEB7: B3, B7, B20,
LTEB8: B1,
LTEB12: B2, B4, B30,
LTEB13: B2, B4,
LTEB20: B3, B7,
LTEB25:
LTEB26:
LTEB29:
LTEB30: B5, B12,
LTEB41: B41,

Permitted Bands:
LTEB1: B8,
LTEB2: B2, B5, B12, B13, B29,
LTEB3: B7, B20,
LTEB4: B4, B5, B12, B13, B29,
LTEB5: B2, B4,
LTEB7: B3, B7, B20,
LTEB8: B1,
LTEB12: B2, B4,
LTEB13: B2, B4,
LTEB20: B3, B7,
LTEB25:
LTEB26:
LTEB29:
LTEB41: B41,

Prune_ca_combos:
Empty

4IceG, 2023-12-16:
Siemanko :) ,

@przemekp1 zmieniles sobie kombinacje pasm na B3+B7+B20, modem lapie coś więcej niż B20?

przemekp1, 2023-12-16:
nie, pracuje tylko na B20 i całkiem nieźle
mogę go wrzucić na auto at!band=00
poczytałem trochę o tym cudaku na tej stronie https://wirelessjoint.com/viewtopic.php?t=664
dzięki tobie zaczynam kumać o co chodzi :)

4IceG, 2023-12-16:
@przemekp1 moze warto to sprawdzić, wrocic do domyslnych pasm/auto. Nie wiem jakie pasma miales agregowane wczesniej czy bylo coś więcej niż B20.

Tylko głośno mysle..
Moze lepiej to sprawdzić żeby się nie okazalo, że np ustawienie niepoprawnej kombinacji pasm coś zawiesza.

przemekp1, 2023-12-16:
poprzednio skakło między B1 a B20 - pewnie zasięg ma tutaj znaczenie, sprawdzę i dam znać

przemekp1, 2023-12-16:
ej sorki, jak dałem na auto to B3 się ładuje, albo skacze na B7 i jak widzę to trzeba zrobić restart interfejsu 
[image]

przemekp1, 2023-12-16:
albo samo jest B7 albo B20 z B3 albo samo B20 - tak skacze

przemekp1, 2023-12-16:
B20 - chyba najbardziej stabilnie - niestety mieszkam na wiosce pod miastem

4IceG, 2023-12-16:
Z dostepnej listy wariantow konfiguracji powinieneś @przemekp1 ustawić B3+B7+B20 inna sprawa jak to się będzie u ciebie sprawdzać w praktyce (stabilnosc/prędkość). Wiadomo że jak ktoś mieszka dalej od nadajnika to B20 będzie miał najmocniejsze.

closed 2023-12-16 by 4IceG