4IceG/luci-app-modemband #13: Support for Quectel RM500Q-GL

atsrxL, 2023-08-30:
Hello， Please add support for Quectel RM500Q-GL

Here's the information you need that u listed:
MODEL:Quectel RM500Q-GL
Revision: RM500QGLABR11A06M4G

USB Device:
Bus 006 Device 003: ID 2c7c:0800 Quectel RM500Q-GL

Serial :
/dev/ttyUSB1

AT command to read current nr5g band setup;
AT+QNWPREFCFG="nr5g_band"
+QNWPREFCFG: "nr5g_band",1:2:3:5:7:8:12:20:25:28:38:40:41:48:66:71:77:78:79

AT command to set nr5g band;
AT+QNWPREFCFG="nr5g_band",41:79

Supported bands;
AT+QGETCAPABILITY?
+QGETCAPABILITY:NR:n1,n2,n3,n5,n7,n8,n12,n20,n25,n28,n38,n40,n41,n48,n66,n71,n77,n78,n79
+QGETCAPABILITY:LTE-FDD:B1,B2,B3,B4,B5,B7,B8,B12,B13,B14,B17,B18,B19,B20,B25,B26,B28,B29,B30,B32,B66,B71
+QGETCAPABILITY:LTE-TDD:B34,B38,B39,B40,B41,B42,B43,B46,B48
+QGETCAPABILITY:WCDMA:B1,B2,B3,B4,B5,B6,B8,B19
+QGETCAPABILITY:LTE-CATEGORY:CAT16
+QGETCAPABILITY:LTE-CA:1

If there is any other info you need ,  I will be glad to provide.
Thank you very much.

4IceG, 2023-08-30:
Hi @atsrxL OK thanks. 

It's funny because the modem probably has the same ID as Quectel RG502Q-EA, so need to identify them somehow.

atsrxL, 2023-08-30:
Hi thank you for your reply , as my unstanding  r m/g 500q-XX & 502q-XX might shares a same pid:vid , can we just set a menu that select model manually? or maybe use ATI returned model name to do the identification.

4IceG, 2023-08-30:
I'll figure something out, I just need some free time.

closed 2023-09-12 by 4IceG

atsrxL, 2023-09-12:
I saw the update, Very thankful to you.

atsrxL, 2023-09-13:
Hi man, There have an another problem i have faced , when I compiled your
3ginfo&modemband into openwrt firmware through menuconfig , or just opkg
install the ipks you've provided , In packet manager shows its installed ,
also sms-tooI is useful, but can't find the entry from luci interface, am
i missing something?

When I try to manually input the address for 3ginfo ,it will shows :

No page is registered at '/admin/modem/3ginfo-lite'.

4IceG, 2023-09-13:
> Hi man, There have an another problem i have faced , when I compiled your 3ginfo&modemband into openwrt firmware through menuconfig , or just opkg install the ipks you've provided , In packet [...]

Hi @atsrxL ,
Describe what exactly you are doing because I don't know if you have an error when compiling or installing packages from ipk.

The packages work and must be visible from the LuCI level. If the package is not visible in the menu (applies to installations with ipk), you need to log out and log in again to the router / clear the browser cache.

4IceG, 2023-09-13:
Send everything again, no images were sent.

atsrxL, 2023-09-13:
Hi , will this work?