4IceG/luci-app-modemband #14: 5G bands locking not working in 1.0.19-20230910 on RM520N-GL

brian4286, 2023-09-11:
The LTE bands are working properly. When you change one or more 5G bands, I briefly see the updated bands in the "Currently set 5G bands", I have the modem rebooting automatically on modification, I can verify the modem reboots. 

Once the modem is back online none of the bands have actually been removed.

4IceG, 2023-09-11:
OK thanks. I see now where my mistake is, I will fix it in the next version.

closed 2023-09-11 by 4IceG

4IceG, 2023-09-12:
Hi @brian4286 , 
Package update is now available for download.

brian4286, 2023-09-12:
I updated to 1.0.20-20230912 and I am still experiencing the same issue. Anything I can do to help debug? The modem is a RM520N-GL.

4IceG, 2023-09-12:
Hi @brian4286 ,
I have the same modem and I removed the package from the router and installed it again from my github and everything works for me.

@brian4286 did you clear your browser cache after reinstalling the package?

You can check if the modemband package works from the console:
```
modemband.sh setbands5g "1 3 7"
modemband.sh getbands5g
```

4IceG, 2023-09-13:
Hi @brian4286 ,
The situation has changed a bit, the modemband package will be modified. Wait for new releases, an update should fix all the problems.

brian4286, 2023-09-14:
@4IceG I noticed a new package was pushed, looks like it working properly for me. 

Just a UI nuance in LUCI, consider updating LTE from "LTE bands" to "Currently set LTE bands". It makes far more sense with the new verbiage in 5G.

4IceG, 2023-09-14:
Hi @brian4286 , as far as I know, the LTE tab looks like this

https://github.com/4IceG/Personal_data/blob/master/zrzuty/1.0.21a.png?raw=true

brian4286, 2023-09-14:
I can open a new issue, I know this no relationship to this issue.

Mine says "LTE bands", I checked and I have English translation and/or auto on in 1.0.21-20230914.

4IceG, 2023-09-14:
The browser cache probably needs to be cleared.

brian4286, 2023-09-14:
100% one browser is caching. Others show the updated heading. Thank you, sorry about that!