4IceG/luci-app-modemband #16: Unknown package 'sms-tool'.

MateuszCOMP, 2024-01-08:
Próbuję zainstalować to na :
> Hostname | OpenWrt
> Model | ZTE MF286D
> Architecture | ARMv7 Processor rev 5 (v7l)
> Target Platform | ipq40xx/generic
> Firmware Version | OpenWrt 22.03.0-rc5 r19523-bfd070e7fa / LuCI openwrt-22.03 branch git-22.167.28394-8a4486a
> root@openwrt:~# picocom -b 115200 /dev/ttyUSB2 : ATI : (BD_TELIAMF286DV1.0.0B12)

Instalacja siada już na pierwszej komendzie:

```
root@OpenWrt:~# opkg install kmod-usb-serial kmod-usb-serial-option sms-tool
Package kmod-usb-serial (5.10.127-1) installed in root is up to date.
Package kmod-usb-serial-option (5.10.127-1) installed in root is up to date.
Unknown package 'sms-tool'.
Collected errors:
 * opkg_install_cmd: Cannot install package sms-tool.
root@OpenWrt:~#
```

4IceG, 2024-01-09:
Cześć @MateuszCOMP ,

Wykonaj w konsoli:

grep -q IceG_repo /etc/opkg/customfeeds.conf || echo 'src/gz IceG_repo https://github.com/4IceG/Modem-extras/raw/main/myrepo' >> /etc/opkg/customfeeds.conf
wget https://github.com/4IceG/Modem-extras/raw/main/myrepo/IceG-repo.pub -O /tmp/IceG-repo.pub
opkg-key add /tmp/IceG-repo.pub
opkg update
opkg install luci-app-modemband

Jeżeli masz wszystko ok w systemie i nic nie namieszales to powinno się zainstalowac.

MateuszCOMP, 2024-01-09:
Mam świeżo postawiony firmware B12. Jedyne co zmieniłem to hasło i ip adress na 192.168.2.1 aby mieć dostęp do interentu.
Pomimo wykonania tych komend, 'luci-app-modemband nie chce się załadować

> **root@OpenWrt**:~# grep -q IceG_repo /etc/opkg/customfeeds.conf || echo 'src/gz Ice
> G_repo https://github.com/4IceG/Modem-extras/raw/main/myrepo' >> /etc/opkg/custo
> mfeeds.conf
> **root@OpenWrt**:~# wget https://github.com/4IceG/Modem-extras/raw/main/myrepo/IceG-
> repo.pub -O /tmp/IceG-repo.pub
> Downloading 'https://github.com/4IceG/Modem-extras/raw/main/myrepo/IceG-repo.pub                                                                                        '
> Connecting to 140.82.121.4:443
> Redirected to /4IceG/Modem-extras/main/myrepo/IceG-repo.pub on raw.githubusercon                                                                                        tent.com
> Writing to '/tmp/IceG-repo.pub'
> /tmp/IceG-repo.pub   100% |*******************************|    92   0:00:00 ETA
> Download completed (92 bytes)
> **root@OpenWrt**:~# opkg-key add /tmp/IceG-repo.pub
> **root@OpenWrt**:~# opkg update
> Downloading https://github.com/4IceG/Modem-extras/raw/main/myrepo/Packages.gz
> Updated list of available packages in /var/opkg-lists/IceG_repo
> Downloading https://github.com/4IceG/Modem-extras/raw/main/myrepo/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/targets/ipq40xx/g                                                                                        eneric/packages/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_core
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/targets/ipq40xx/g                                                                                        eneric/packages/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cort                                                                                        ex-a7_neon-vfpv4/base/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_base
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cort                                                                                        ex-a7_neon-vfpv4/base/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cort                                                                                        ex-a7_neon-vfpv4/luci/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_luci
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cort                                                                                        ex-a7_neon-vfpv4/luci/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cort                                                                                        ex-a7_neon-vfpv4/packages/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_packages
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cort                                                                                        ex-a7_neon-vfpv4/packages/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_routing
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_telephony
> Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.sig
> Signature check passed.
> **root@OpenWrt**:~# opkg install luci-app-modemband
> Unknown package 'luci-app-modemband'.
> Collected errors:
>  * pkg_hash_check_unresolved: cannot find dependency sms-tool for modemband
>  * pkg_hash_fetch_best_installation_candidate: Packages for modemband found, but incompatible with the architectures configured
>  * pkg_hash_check_unresolved: cannot find dependency sms-tool for luci-app-modemband
>  * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-modemband found, but incompatible with the architectures configured
>  * opkg_install_cmd: Cannot install package luci-app-modemband.
> **root@OpenWrt**:~#

4IceG, 2024-01-09:
@MateuszCOMP używasz starszej, nieaktualnej wersji OpenWrt, nie pamiętam czy w tym wydaniu był już dodany sms-tool.

Wgraj sobie aktualne wydanie [Pobierz](https://downloads.openwrt.org/releases/23.05.2/targets/ipq40xx/generic/openwrt-23.05.2-ipq40xx-generic-zte_mf286d-squashfs-sysupgrade.bin)

lub od Cezarego [Pobierz](https://dl.eko.one.pl/luci/openwrt-23.05/ipq40xx/generic/luci-23.05-snapshot-r23695-80ef582dee-ipq40xx-generic-zte_mf286d-squashfs-sysupgrade.bin) [Zalecane jeżeli chcesz mieć łatwiej w konfiguracji]

Po aktualizacji wiadomo ponownie zmień sobie IP, skonfiguruj qmi aby mieć neta, ustaw hasło. Po tych krokach wykonaj polecenia, które podałem wyżej.

MateuszCOMP, 2024-01-09:
Zainstalowałem wersję od cezarego
Jest obecnie wersja firmware：OpenWrt 23.05-SNAPSHOT r23695-80ef582dee

Jednak finalnie dalej dostaję błąd:
_Unknown package 'luci-app-modemband'.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-app-modemband._

Tutaj są komendy których użyłem
```
root@OpenWrt:~# opkg install kmod-usb-serial kmod-usb-serial-option sms-tool
Package kmod-usb-serial (5.15.139-1) installed in root is up to date.
Package kmod-usb-serial-option (5.15.139-1) installed in root is up to date.
Package sms-tool (2023-09-21-1b6ca032-1) installed in root is up to date.
```
```
root@OpenWrt:~# opkg update
Downloading https://github.com/4IceG/Modem-extras/raw/main/myrepo/Packages.gz
*** Failed to download the package list from https://github.com/4IceG/Modem-extras/raw/main/myrepo/Packages.gz

Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz

Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz

Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz

Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz

Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz

Downloading https://dl.eko.one.pl/openwrt-23.05/targets/ipq40xx/generic/packages/Packages.gz
*** Failed to download the package list from https://dl.eko.one.pl/openwrt-23.05/targets/ipq40xx/generic/packages/Packages.gz

Downloading https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz
*** Failed to download the package list from https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz

Downloading https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz
*** Failed to download the package list from https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz

Downloading https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
*** Failed to download the package list from https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz

Downloading https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
*** Failed to download the package list from https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz

Downloading https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
*** Failed to download the package list from https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz

Downloading https://dl.eko.one.pl/openwrt-23.05/targets/ipq40xx/generic/kmods-5.15.139/Packages.gz
*** Failed to download the package list from https://dl.eko.one.pl/openwrt-23.05/targets/ipq40xx/generic/kmods-5.15.139/Packages.gz

Collected errors:
 * opkg_download: Failed to download https://github.com/4IceG/Modem-extras/raw/main/myrepo/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://dl.eko.one.pl/openwrt-23.05/targets/ipq40xx/generic/packages/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz, wget returned 1.
[... cut ...]
 * opkg_download: Failed to download https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://dl.eko.one.pl/openwrt-23.05/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://dl.eko.one.pl/openwrt-23.05/targets/ipq40xx/generic/kmods-5.15.139/Packages.gz, wget returned 1.
root@OpenWrt:~# opkg install luci-app-modemband
Unknown package 'luci-app-modemband'.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-app-modemband.
root@OpenWrt:~#
```

4IceG, 2024-01-09:
@MateuszCOMP nie wiem co robisz konfigurując router ale to nie tak powinno wyglądać, masz pomerdane repozytoria i jeszcze nie pobiera paczek tak jakby neta nie było.

Instalując obraz od Cezarego masz mieć pakiety pobierane tylko z https://dl.eko.one.pl/openwrt-23.05/, a u ciebie jest mix wydanie oficjalne + Cezarego.

Przywróć router do ustawień domyślnych, nie zachowuj konfiguracji i ustaw wszystko krok po kroku.

4IceG, 2024-01-09:
Nie musi być dwa razy instalowane sms-tool. Jeżeli masz obraz od Cezarego to pakiety kmod-usb-serial oraz kmod-usb-serial-option powinny już być wgrane (ale mogę się mylić bo dawno nie używałem obrazów z eko.one.pl), więc powinno wystarczyć tyle:

```
grep -q IceG_repo /etc/opkg/customfeeds.conf || echo 'src/gz IceG_repo https://github.com/4IceG/Modem-extras/raw/main/myrepo' >> /etc/opkg/customfeeds.conf
wget https://github.com/4IceG/Modem-extras/raw/main/myrepo/IceG-repo.pub -O /tmp/IceG-repo.pub
opkg-key add /tmp/IceG-repo.pub
opkg update
opkg install luci-app-modemband
```

MateuszCOMP, 2024-01-09:
Zrobiłem reset urządzenia.
Po resecie jedyną czynnością jaką zrobiłem w web była zmiana adresacji routera na 192.168.2.1 aby uzyskał dostęp do internetu po moście.
Potem już Putty i tutaj pełny log. Chyba wygląda wszystko dobrze?

4IceG, 2024-01-09:
@MateuszCOMP jest OK. Tak powinno to wyglądać.

Polecam również doinstalować luci-app-3ginfo-lite i luci-app-sms-tool-js (mogą się też przydać).

closed 2024-01-09 by 4IceG

MateuszCOMP, 2024-01-09:
Sprawdziłem wszystko gra, dodatkowa zakładka się pojawiła.
Dzięki za pomoc 👍