4IceG/luci-app-modemband #18: Support on Quectel RM500U-CNV

flammable47, 2024-01-28:
exact name of modem and ew version number:
Quectel
RM500U-CNV
Revision: RM500UCNVAAR03A02M2G

VID and PID identifier on the USB bus:
Bus 002 Device 003: ID 2c7c:0900 Quectel RM500U-CNV

serial port used for communication with the modem ("diagnostic"):
/dev/ttyUSB2

AT command to read the set bands together with an example result:
AT+QNWINFO
+QNWINFO: "NR5G-SA",51566,"NR N78",620640

an AT command to set specific bands

4IceG, 2024-01-28:
Hi,
@flammable47 Check if it works, I also added modem to 3ginfo-lite package (you can also test it).

[modemband.zip](https://github.com/4IceG/luci-app-modemband/files/14075513/modemband.zip)
[3ginfo-lite.zip](https://github.com/4IceG/luci-app-modemband/files/14075520/3ginfo-lite.zip)

flammable47, 2024-01-29:
> Hi, @flammable47 Check if it works, I also added modem to 3ginfo-lite package (you can also test it).
> 
> [modemband.zip](https://github.com/4IceG/luci-app-modemband/files/14075513/modemband.zip) [3ginfo-lite.zip](https://github.com/4IceG/luci-app-modemband/files/14075520/3ginfo-lite.zip)

Hello @4IceG 
I already check the package and it's works great..!!
[image]
[image]
[image]
[image]

4IceG, 2024-01-29:
Hi,
Unfortunately, it does not show all data and I do not know if it's the fault of the "ecm protocol" or if you modified the settings in modem. It should return the 5GNR NSA bands (they are completely missing from the settings), 3ginfo-lite does not show the data correctly either.

@flammable47 Can you restore the modem to default settings and run it on the qmi/mbim protocol?

flammable47, 2024-01-29:
> Hi, Unfortunately, it does not show all data and I do not know if it's the fault of the "ecm protocol" or if you modified the settings in modem. It should return the 5GNR NSA bands (they are completely missing from the settings), 3ginfo-lite does not show the data correctly either.
> 
> @flammable47 Can you restore modem to default settings and run it on the qmi/mbim protocol?

It's only supported in ECM protocol or it can be NDIS, RNDIS, NCM?

4IceG, 2024-01-29:
OK, I understand, let's keep this "ecm".

@flammable47 Show command result:
`sms_tool -d /dev/ttyUSB2 at "at+qnwinfo;+qcainfo;+qeng=\"servingcell\";+qspn"`

maybe we will find out why 3ginfo-lite does not show data.

flammable47, 2024-01-29:
> OK, I understand, let's keep this "ecm".
> 
> @flammable47 Show command result: `sms_tool -d /dev/ttyUSB2 at "at+qnwinfo;+qcainfo;+qeng=\"servingcell\";+qspn"`
> 
> maybe we will find out why 3ginfo-lite does not show data.

here the result of the script
[image]

4IceG, 2024-01-29:
I don't know if it's OK now, unusual modem, strange AT commands.

[3ginfo-lite.zip](https://github.com/4IceG/luci-app-modemband/files/14085005/3ginfo-lite.zip)

closed 2024-01-29 by 4IceG