4IceG/luci-app-modemband #19: Is it possible to add a cell lock function?

OvidioLee, 2024-03-14:
The plugin you created is perfect, but could you add a cell lock function.

First get the frequency information from at, then choose one for cell lock.

4IceG, 2024-03-14:
Hi @OvidioLee ,
This functionality will not be included in this package. If I were to add something like this, it would be as a new tab for 3ginfo-lite.

OvidioLee, 2024-03-17:
I highly recommend you add this feature.
new tab for 3ginfo-lite. its ok

cariad61, 2024-03-22:
Thank you for all your packages they all installed and worked first time.

For information only I get the following warning/error when installing some packages:

Configuring` modemband.
Configuring luci-app-modemband.

`uci: Parse error (invalid character in name field) at line 49, byte 20`

My system info:

<h3>System</h3><div>
Hostname | TarbatMainOpenWrt
-- | --
Model | Techvision TVI7309X
Architecture | Intel(R) Celeron(R) N5100 @ 1.10GHz
Target Platform | x86/64
Firmware Version | OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-23.306.39416-c86c256
Kernel Version | 5.15.137
Local Time | 2024-03-22 13:06:16
Uptime | 14h 43m 19s

</div>

Thank you for all the information you provide about the Quectel modems.

You probably know already but the command:

`AT+QCFG="lte/bandprior",20,3`

is very useful to influence which is the primary channel for some CA combinations.

4IceG, 2024-03-23:
> Thank you for all your packages they all installed and worked first time.
> 
> For information only I get the following warning/error when installing some packages:
> 
> Configuring` modemband. Configuring luci-app-modemband.
> 
> `uci: Parse error (invalid character in name field) at line 49, byte 20`

Hi @cariad61 
I wrote packages for myself, as an ordinary OpenWrt user, not a developer.
All this means that errors may appear.

kokroo, 2024-05-24:
@4IceG The primary band lock feature would be quite useful. Is it available in this package, or another package (even yours)?

Cheers

obsy, 2024-05-25:
Nie wiem czy to powinno należeć do modemband. Ja bym to widział raczej w 3ginfo w gui, na zasadzie - jak wiesz że dany modem to potrafi i masz do niego odpowiednie polecenie, jak masz w parametrach PCI i EARFCN to wtedy dajesz jakiśś przycisk który jest aktywny i klikasz go żeby modem zbindował się na te parametry. Ew ręcznie wprowadzasz PCI i EARFCN gdzieś tam w konfiguracji.

4IceG, 2024-05-25:
@obsy Cezary opcja "cell lock" zgodze się z Tobą to bardziej zadanie dla 3ginfo. Dla modemband chodzilo o dodanie funkcji priorytetu dla pasm.

Manu99it, 2024-06-04:
Yeah would be interesting to see somewhere a cell lock setting. For now I am using AT commands, at least on many quectel modems to retrieve cell info one can use AT+QENG="neighbourcell"  and to lock cell 
AT+QNWLOCK="common/4g",<number of cell>,<erfcn>,<PCI>,<erfcn2>,<pci2>,etc...
To disable cell lock command is AT+QNWLOCK="common/4g",0

Would also be nice to have a setting to set 4g only mode or 4g/WCDMA/UMTS

4IceG, 2024-06-09:
To add new function I need to rewrite the 3ginfo-lite plugin. I don't know if I can do it quickly.