4IceG/luci-app-modemband #25: Add Yuge CLM920 NC_5 Support

mazige1997, 2024-06-17:
hi my friend @4IceG,
are you able to add to the custom band list also Yuge CLM920 NC_5 ?
[Here](https://docs.yuge-info.com/lib/exe/fetch.php?media=9x07:9x07_at_commands_manual_v1.16.pdf) is the at commands manual

mazige1997, 2024-06-17:
It's vid is 05c6.
It's pid is 9025.
Support LTE_BANDS: "1 3 5 8 38 39 40 41"

4IceG, 2024-06-17:
Hi,
@mazige1997 I don't promise anything, but We can try. 
P.S.
Does modem have a communication port?

obsy, 2024-06-18:
@mazige1997 Can be done, just provide the communication port.

4IceG, 2024-06-19:
@obsy Cezary strzelam, że port do komunikacji to ttyUSB2. Co do ID modemu nie jestem pewien https://www.spinics.net/lists/linux-usb/msg164213.html & https://lore.kernel.org/all/lsq.1519831221.18292864@decadent.org.uk/T/.

mazige1997, 2024-06-19:
> @mazige1997 Can be done, just provide the communication port.

The  at communication port is /dev/ttyUSB2

4IceG, 2024-06-19:
@mazige1997 Show Us the output of the command `cat /sys/kernel/debug/usb/devices` for this modem.

mazige1997, 2024-06-19:
> @mazige1997 Show Us the output of the command `cat /sys/kernel/debug/usb/devices` for this modem.

root@ImmortalWrt:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.01
S:  Manufacturer=Linux 6.1.92-ophub ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=fe380000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=05c6 ProdID=9625 Rev= 3.18
S:  Manufacturer=Shanghai Yuge
S:  Product=YUGA Incorporated
S:  SerialNumber=0123456789ABCDEF
C:* #Ifs= 5 Cfg#= 1 Atr=a0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=ff Driver=option
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=42 Prot=01 Driver=(none)
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=84(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=83(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 3 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=86(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=85(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=qmi_wwan
E:  Ad=88(I) Atr=03(Int.) MxPS=   8 Ivl=32ms
E:  Ad=87(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=05(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.01
S:  Manufacturer=Linux 6.1.92-ophub ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=fe3c0000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 7
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=02 MxPS=64 #Cfgs=  1
P:  Vendor=1a40 ProdID=0201 Rev= 1.00
S:  Product=USB 2.0 Hub [MTT]
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=01 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms
I:* If#= 0 Alt= 1 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=02 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms

T:  Bus=03 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=12   MxCh= 1
B:  Alloc=  0/900 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 1.10 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0001 Rev= 6.01
S:  Manufacturer=Linux 6.1.92-ophub ohci_hcd
S:  Product=Generic Platform OHCI controller
S:  SerialNumber=fe3a0000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   2 Ivl=255ms

T:  Bus=04 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=12   MxCh= 1
B:  Alloc=  0/900 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 1.10 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0001 Rev= 6.01
S:  Manufacturer=Linux 6.1.92-ophub ohci_hcd
S:  Product=Generic Platform OHCI controller
S:  SerialNumber=fe3e0000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   2 Ivl=255ms

T:  Bus=05 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.01
S:  Manufacturer=Linux 6.1.92-ophub xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.4.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=06 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 6.01
S:  Manufacturer=Linux 6.1.92-ophub xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.4.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=07 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.01
S:  Manufacturer=Linux 6.1.92-ophub xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.5.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=08 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 6.01
S:  Manufacturer=Linux 6.1.92-ophub xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.5.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

obsy, 2024-06-19:
9625, not 9025

mazige1997, 2024-06-19:
> 9625, not 9025

Sorry, just type error.

mazige1997, 2024-06-20:
> 9625, not 9025
[Here](http://www.lc-chip.com/upfile/img/221023/635436fc90894.jpg)  and [here](http://www.lc-chip.com/upfile/img/221023/635436fcbc813.jpg) is the support band !

obsy, 2024-06-20:
https://github.com/obsy/packages/commit/307cd05fa9f81bdb331daf90c9bf7bf24c1b9a9c

mazige1997, 2024-06-20:
> obsy/packages@307cd05

Thanks very much !

mazige1997, 2024-06-20:
> Hi, @mazige1997 I don't promise anything, but We can try. P.S. Does modem have a communication port?
Here is commit,  can you add clm920 nc_5 support to luci-app-modemband and luci-app-3ginfo-lite?

4IceG, 2024-06-20:
> Here is commit, can you add clm920 nc_5 support to luci-app-modemband and luci-app-3ginfo-lite?

@mazige1997 Extract archive [test_ipk.zip](https://github.com/user-attachments/files/15920027/test_ipk.zip) and check packages.

mazige1997, 2024-06-21:
> > [obsy/packages@307cd05](https://github.com/obsy/packages/commit/307cd05)
> 
> Thanks very much !

It work well. Thanks!

mazige1997, 2024-06-21:
> > Here is commit, can you add clm920 nc_5 support to luci-app-modemband and luci-app-3ginfo-lite?
> 
> @mazige1997 Extract archive [test_ipk.zip](https://github.com/user-attachments/files/15920027/test_ipk.zip) and check packages.

It work well. Thanks!

closed 2024-06-23 by 4IceG