4IceG/luci-app-modemband #26: Working with modemmanager

Manu99it, 2024-06-21:
In the last two days for an unrelated problem (frequent disconnections) I installed modemmanager. For curiosity I tried your packages and all worked without serious problems. Just sometimes they couldn't retrieve values from modem, but simply reloading the page made it work. I was thinking that if the problem is the concurrent usage of qmi by modemmanager and your packages maybe you can use qmi-proxy of libqmi

4IceG, 2024-06-21:
Hi,
@Manu99it Sorry, but there is information in package description not to use my packages with MM. Multiple applications cannot use modem at the same time.

I will not fight with MM and specially modify packages. Create a package for LuCI that supports all MM functions, my packages will then be unnecessary.

Manu99it, 2024-06-21:
> Multiple applications cannot use modem at the same time.

Yes but for example if one keep installed uqmi and modemmanager, these go effectively in conflict with modemmanager giving an error after some times that the modem can't be found. What I'm saying is that after uninstalling uqmi, your packages continue to work and modemmanager seems to not bother

closed 2024-06-21 by Manu99it