4IceG/luci-app-modemband #37: Fibocom FM350 the bands return default on restart

SuperKali, 2024-11-30:
Hi,

I encountered a strange anomaly, when I configure the allowed bands, on the next restart these are set by default to all bands.

How is this logic handled?

Is it possible to solve it?

Thank you

4IceG, 2024-11-30:
Hi,
@SuperKali unpack archive [_fibocom_fm350_common.zip](https://github.com/user-attachments/files/17967280/_fibocom_fm350_common.zip) and copy file to `/usr/share/modemband/`. Check if it's OK.

SuperKali, 2024-11-30:
Hi,

I did what you told me but the problem still arises, I read the AT manual, it seems that the setband is not persistent on the module side:
A video example:
https://share.superkali.me/s/wdCERfexFaTEfR6

I hope it can be fixed 

Thank you.

4IceG, 2024-12-08:
@prusa-dev does your modem remember the modified bands? Because I don't know if it's a problem with modem or if scripts need to be modified.

SuperKali, 2024-12-08:
@4IceG I read the AT manual and the changes are not persistent

SuperKali, 2024-12-08:
[image]