4IceG/luci-app-modemband #38: Unable to install sms-tool on GL.iNet MT3000

Bonnyfede00, 2024-12-30:
i’m trying to install sms-tool on my GL.iNet MT3000 running OpenWrt 21.02, but it fails with the following error: 
Unknown package 'sms-tool'.
Collected errors:
 * opkg_install_cmd: Cannot install package sms-tool.
I've added the correct repository for my architecture, and the only similar package is called 'smstools3'. How can I resolve this? I will be using an FM350GL.
Thank you so much.

4IceG, 2024-12-30:
Hi,
Update OpenWrt (recommended) or try to install package manually. The sms-tool is available in the OpenWrt repository since v23.05.

Bonnyfede00, 2025-01-02:
Ok, now I managed to install it, my FM350-GL modem is detected, but I can't get it to connect, is there any other step I need to take?