4IceG/luci-app-modemband #41: ZLT S12 Pro

flammable47, 2025-04-25:
Good day

I would like to request if this is possible to support:

Bus 001 Device 002: ID 1782:4055 Unisoc Phone Unisoc Phone
/dev/ttyUSB0

as of today only this command that been applied on the router
AT+CREG?
AT+CREG=?
AT+CGREG?
AT+CGREG=?
AT+GMI=?
AT+GMI
AT+GMM=?

[image]
[image]
[image]
[image]

flammable47, 2025-04-28:
> Hi, Not enough information to add modem support. The most important thing is missing.. at commands to change bands and interface for communication with modem.
> 
> > UNISOC vendor ID and TOZED LT70-C modem which is based from UNISOC
> > SL8563. The modem supports the NCM mode. Interface 0 is used for running
> > the AT commands. Interface 12 is the ADB interface.
> > P:  Vendor=1782 ProdID=4055 Rev=04.04
> > S:  Manufacturer=Unisoc Phone
> > S:  Product=Unisoc Phone

is these the following command is needed?

[image]

ChamodyaChiran, 2025-11-09:
/dev/ttyUSB0

get band-AT+SPLBAND=0
set band-AT+SPLBAND=<mode>,<band_tdd_49_64>,
<band_tdd_33_48>,<band_fdd_17_32>,<band
_fdd_1_16>,< band_fdd_65_80>

I have attached pdf

[band set.pdf](https://github.com/user-attachments/files/23440670/band.set.pdf)